CommonWealthRobotics/CaDoodle-Application #11: Missing square to move single dimension

JansenSmith, 2024-10-26:
Attempting to move the rectangle into the circles. No interface exists to do so easily. Please address.
[image]

madhephaestus, 2025-01-05:
When implemented: holding shift should perform a uniform scaling operation.

madhephaestus, 2025-12-03:
@JansenSmith was working on this and had a similar issue. There are 2 paths forward here. First (and easiest) is to implement the uniform scaling on the dimensions instead/addition to the drag handle. that would let you hold shift while clicking on the handle, then use the number fields to set one of the sides to set the scale. 

the way to correct the issue is a bit awkward. You can see that the drag event is the event that performs the scale operation. Once the uniform scaling is started, you need to ignore any of the drag events from the corner itself. You are getting a loop when the event that receives the mouse move, causes the corner to move, causing another mouse move event, and that loop continues until the corner is zoomed away. You need to look at the event source for the mouse listener, and when the mouse listener is in uniform scale mode, and already started dragging, only use mouse events that are not sourced from the corner mesh.

JansenSmith, 2025-12-03:
Oh, @rondlh . Here is the PR I have trying to do exactly this: https://github.com/CommonWealthRobotics/CaDoodle-Application/pull/96