CommonWealthRobotics/CaDoodle-Application #5: Crashed - Stack trace

JansenSmith, 2024-10-26:
java.io.FileNotFoundException: /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/UnKnown (/home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/UnKnown)
Saved ConfigurationDatabase.json
	at groovy.lang.GroovyCodeSource.<init>(GroovyCodeSource.java:107)
Placing Star at 5 , 2
	at groovy.lang.GroovyShell.parse(GroovyShell.java:606)
Saved ConfigurationDatabase.json
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:54)
Placing icosahedron at 6 , 0
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:78)
Saved ConfigurationDatabase.json
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:521)
Placing Adjustable Star at 6 , 1
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:1114)
Saved ConfigurationDatabase.json
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.flaten(ScriptingEngine.java:134)
Placing Diamond at 6 , 2
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.AddFromFile.process(AddFromFile.java:54)
Saved ConfigurationDatabase.json
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.process(CaDoodleFile.java:211)
Saved ConfigurationDatabase.json
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.initialize(CaDoodleFile.java:123)
Screen 2256.0x1504.0
	at com.commonwealthrobotics.MainController.lambda$setupFile$9(MainController.java:685)
Current opperation results: Add Object
	at java.base/java.lang.Thread.run(Thread.java:840)
Adding file to listening /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/UnKnown
Auto save /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/Doodle-0.doodle

JansenSmith, 2024-10-26:
[SecretMessageCoin.groovy.zip](https://github.com/user-attachments/files/17531021/SecretMessageCoin.groovy.zip)

madhephaestus, 2024-10-26:
see https://github.com/CommonWealthRobotics/CaDoodle-Application/actions/runs/11532831155/job/32105056602

JansenSmith, 2024-10-26:
crashed again

JansenSmith, 2024-10-26:
Splash Rendering 50 Add Object running
Found 3 mesh objects
Object: coin3254709524566077489.stl4238049978210375957, Faces: 25577
Object: reds4660387347979582551.stl1298088782947950892, Faces: 2304
Object: blacks8080948558096992969.stl7266252917793739639, Faces: 1472
Blender file loaded from: /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-1/Zippy_Exosuit_3bf04gFGMr.blend
ASCII STL exported to: /tmp/Zippy_Exosuit_3bf04gFGMr.blend10948067420767869869.stl
STL file size: 1467734 bytes
Blender 4.2.3 LTS (hash 0e22e4fcea03 built 2024-10-14 23:31:34)
Read blend: "/home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-1/Zippy_Exosuit_3bf04gFGMr.blend"
Timer 'STL Export' took 2.98 ms

Blender quit

java.lang.NumberFormatException: This set of points is not a valid polygon
	at eu.mihosoft.vrl.v3d.Plane.computeNormal(Plane.java:144)
	at eu.mihosoft.vrl.v3d.Plane.createFromPoints(Plane.java:102)
	at eu.mihosoft.vrl.v3d.Polygon.<init>(Polygon.java:115)
	at eu.mihosoft.vrl.v3d.Polygon.fromPoints(Polygon.java:451)
	at eu.mihosoft.vrl.v3d.Polygon.fromPointsAllowDegenerate(Polygon.java:427)
	at eu.mihosoft.vrl.v3d.STL.file(STL.java:92)
	at com.neuronrobotics.bowlerstudio.vitamins.Vitamins.get(Vitamins.java:87)
	at com.neuronrobotics.bowlerstudio.scripting.BlenderLoader.scaleStl(BlenderLoader.java:94)
	at com.neuronrobotics.bowlerstudio.scripting.BlenderLoader.toSTLFile(BlenderLoader.java:153)
	at com.neuronrobotics.bowlerstudio.scripting.BlenderLoader.inlineScriptRun(BlenderLoader.java:26)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:521)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:1114)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.flaten(ScriptingEngine.java:134)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.AddFromFile.process(AddFromFile.java:55)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.process(CaDoodleFile.java:211)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.lambda$addOpperation$2(CaDoodleFile.java:240)
	at java.base/java.lang.Thread.run(Thread.java:840)
Initial Loading /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-1/Zippy_Exosuit_3bf04gFGMr.blend
 Splash Rendering 100 Initialize Model
Adding Listeners for Zippy_Exosuit_3bf04gFGMr
Adding file to listening /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-1/Zippy_Exosuit_3bf04gFGMr.blend
Adding listener to Zippy_Exosuit_3bf04gFGMr_CaDoodle_File
Auto save /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-1/Doodle-1.doodle
:ExternalEditorController:  Zippy_Exosuit_3bf04gFGMr.blend is not supported by class com.neuronrobotics.bowlerstudio.scripting.external.SVGExternalEditor
AssetFactory loaded
:ExternalEditorController:  Zippy_Exosuit_3bf04gFGMr.blend is not supported by class com.neuronrobotics.bowlerstudio.scripting.external.GroovyEclipseExternalEditor
No image at Blender.png
:ExternalEditorController:  Zippy_Exosuit_3bf04gFGMr.blend is not supported by class com.neuronrobotics.bowlerstudio.scripting.external.ArduinoExternalEditor
java.lang.RuntimeException
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAsset(AssetFactory.java:111)
	at com.neuronrobotics.bowlerstudio.scripting.external.BlenderExternalEditor.getImage(BlenderExternalEditor.java:101)
	at com.neuronrobotics.bowlerstudio.scripting.external.ExternalEditorController.<init>(ExternalEditorController.java:43)
	at com.commonwealthrobotics.controls.SelectionSession.setUpFileBox(SelectionSession.java:487)
	at com.commonwealthrobotics.controls.SelectionSession.updateSelection(SelectionSession.java:408)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$setUpControls$7(SelectionSession.java:340)
	at javafx.base/com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:232)
	at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:189)
	at javafx.base/com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at javafx.base/com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at javafx.base/com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.base/javafx.event.Event.fireEvent(Event.java:198)
	at javafx.graphics/javafx.scene.Scene$MouseHandler.process(Scene.java:3985)
[... cut ...]
	at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:449)
	at javafx.graphics/com.sun.glass.ui.View.handleMouseEvent(View.java:551)
	at javafx.graphics/com.sun.glass.ui.View.notifyMouse(View.java:937)
	at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$10(GtkApplication.java:264)
	at java.base/java.lang.Thread.run(Thread.java:840)
x`CaDoodle Updater clean exit

JansenSmith, 2024-10-26:
[SecretMessageCoin.groovy.2.zip](https://github.com/user-attachments/files/17531103/SecretMessageCoin.groovy.2.zip)

JansenSmith, 2024-10-26:
Splash Rendering 50 Add Object running
Found 3 mesh objects
Object: coin3254709524566077489.stl4238049978210375957, Faces: 25577
Object: reds4660387347979582551.stl1298088782947950892, Faces: 2304
Object: blacks8080948558096992969.stl7266252917793739639, Faces: 1472
Blender file loaded from: /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/Productive_Hyperloop_McM4pPNgWr.blend
ASCII STL exported to: /tmp/Productive_Hyperloop_McM4pPNgWr.blend5223781224577352253.stl
STL file size: 1467734 bytes
Blender 4.2.3 LTS (hash 0e22e4fcea03 built 2024-10-14 23:31:34)
Read blend: "/home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/Productive_Hyperloop_McM4pPNgWr.blend"
Timer 'STL Export' took 3.00 ms

Blender quit

java.lang.NumberFormatException: This set of points is not a valid polygon
	at eu.mihosoft.vrl.v3d.Plane.computeNormal(Plane.java:144)
	at eu.mihosoft.vrl.v3d.Plane.createFromPoints(Plane.java:102)
	at eu.mihosoft.vrl.v3d.Polygon.<init>(Polygon.java:115)
	at eu.mihosoft.vrl.v3d.Polygon.fromPoints(Polygon.java:451)
	at eu.mihosoft.vrl.v3d.Polygon.fromPointsAllowDegenerate(Polygon.java:427)
	at eu.mihosoft.vrl.v3d.STL.file(STL.java:92)
	at com.neuronrobotics.bowlerstudio.vitamins.Vitamins.get(Vitamins.java:87)
	at com.neuronrobotics.bowlerstudio.scripting.BlenderLoader.scaleStl(BlenderLoader.java:94)
	at com.neuronrobotics.bowlerstudio.scripting.BlenderLoader.toSTLFile(BlenderLoader.java:153)
	at com.neuronrobotics.bowlerstudio.scripting.BlenderLoader.inlineScriptRun(BlenderLoader.java:26)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:521)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:1114)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.flaten(ScriptingEngine.java:134)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.AddFromFile.process(AddFromFile.java:55)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.process(CaDoodleFile.java:211)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.lambda$addOpperation$2(CaDoodleFile.java:240)
	at java.base/java.lang.Thread.run(Thread.java:840)
Initial Loading /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/Productive_Hyperloop_McM4pPNgWr.blend
Adding file to listening /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/Productive_Hyperloop_McM4pPNgWr.blend
Adding Listeners for Productive_Hyperloop_McM4pPNgWr
Adding listener to Productive_Hyperloop_McM4pPNgWr_CaDoodle_File
Auto save /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/Doodle-0.doodle
Allign canceled here
Cancle
Auto save /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-0/Doodle-0.doodle

:ExternalEditorController:  Productive_Hyperloop_McM4pPNgWr.blend is not supported by class com.neuronrobotics.bowlerstudio.scripting.external.SVGExternalEditor
AssetFactory loaded
:ExternalEditorController:  Productive_Hyperloop_McM4pPNgWr.blend is not supported by class com.neuronrobotics.bowlerstudio.scripting.external.GroovyEclipseExternalEditor
No image at Blender.png
:ExternalEditorController:  Productive_Hyperloop_McM4pPNgWr.blend is not supported by class com.neuronrobotics.bowlerstudio.scripting.external.ArduinoExternalEditor
java.lang.RuntimeException
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAsset(AssetFactory.java:111)
	at com.neuronrobotics.bowlerstudio.scripting.external.BlenderExternalEditor.getImage(BlenderExternalEditor.java:101)
	at com.neuronrobotics.bowlerstudio.scripting.external.ExternalEditorController.<init>(ExternalEditorController.java:43)
	at com.commonwealthrobotics.controls.SelectionSession.setUpFileBox(SelectionSession.java:487)
	at com.commonwealthrobotics.controls.SelectionSession.updateSelection(SelectionSession.java:408)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$setUpControls$7(SelectionSession.java:340)
	at javafx.base/com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:232)
	at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:189)
	at javafx.base/com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at javafx.base/com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at javafx.base/com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
[... cut ...]
	at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:449)
	at javafx.graphics/com.sun.glass.ui.View.handleMouseEvent(View.java:551)
	at javafx.graphics/com.sun.glass.ui.View.notifyMouse(View.java:937)
	at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$10(GtkApplication.java:264)
	at java.base/java.lang.Thread.run(Thread.java:840)
CaDoodle Updater clean exit

madhephaestus, 2024-10-26:
clear the cache and run the latest release

commit c7770bca (2024-10-26): close #5
closed 2024-10-26 by madhephaestus