CommonWealthRobotics/CaDoodle-Application #25: Possible mem leak in Group operation

JansenSmith, 2024-11-06:
It's a large STL being grouped, but this machine has more than enough mem for the operation.
[image]

```
Exception in thread "addOpperation Thread 1" java.lang.OutOfMemoryError: Java heap space
Nov 06, 2024 11:15:29 AM sun.awt.X11.XToolkit processException
WARNING: Exception on Toolkit thread
java.lang.OutOfMemoryError: Java heap space
	at java.base/java.util.LinkedList.listIterator(LinkedList.java:872)
	at java.base/java.util.AbstractList.listIterator(AbstractList.java:311)
	at java.base/java.util.AbstractSequentialList.iterator(AbstractSequentialList.java:238)
	at java.desktop/sun.awt.X11.XWM.isStateChange(XWM.java:1219)
	at java.desktop/sun.awt.X11.XFramePeer.handlePropertyNotify(XFramePeer.java:313)
	at java.desktop/sun.awt.X11.XBaseWindow.dispatchEvent(XBaseWindow.java:1173)
	at java.desktop/sun.awt.X11.XBaseWindow.dispatchToWindow(XBaseWindow.java:1116)
	at java.desktop/sun.awt.X11.XToolkit.dispatchEvent(XToolkit.java:616)
	at java.desktop/sun.awt.X11.XToolkit.run(XToolkit.java:731)
	at java.desktop/sun.awt.X11.XToolkit.run(XToolkit.java:647)
	at java.base/java.lang.Thread.run(Thread.java:840)
```

madhephaestus, 2024-11-24:
show me the mem usage graph in visualvm with any memory usage graph please. just the exception is not enough information to do anything with

madhephaestus, 2025-01-07:
@JansenSmith I just made a release that increased the base max ram usage from 8 to 16 gb. Can you try it now?

madhephaestus, 2025-01-08:
this may be solved by https://github.com/CommonWealthRobotics/CaDoodle-Application/commit/d8e87301edbbdec45acba25d28f3001c4d5a93b3

closed 2025-01-08 by madhephaestus