CommonWealthRobotics/CaDoodle-Application #27: Trying to manipulate a large filament painting

JansenSmith, 2024-11-23:
```
Splash Rendering 96 Initialize Model
Exception in thread "AWT-XAWT" java.lang.OutOfMemoryError: Java heap space
java.lang.OutOfMemoryError: Java heap space
java.lang.OutOfMemoryError: Java heap space
java.lang.OutOfMemoryError: Java heap space: failed reallocation of scalar replaced objects
java.lang.Exception: UI Thread Exception here!
	at com.neuronrobotics.bowlerstudio.BowlerStudio.runLater(BowlerStudio.java:536)
	at com.commonwealthrobotics.controls.SelectionSession.displayCurrent(SelectionSession.java:272)
	at com.commonwealthrobotics.controls.SelectionSession.onUpdate(SelectionSession.java:177)
	at com.commonwealthrobotics.MainController.onUpdate(MainController.java:880)
	at com.commonwealthrobotics.ActiveProject.onUpdate(ActiveProject.java:199)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.setCurrentState(CaDoodleFile.java:363)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.process(CaDoodleFile.java:215)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.initialize(CaDoodleFile.java:123)
	at com.commonwealthrobotics.ProjectManager.lambda$loadFiles$2(ProjectManager.java:119)
	at java.base/java.lang.Thread.run(Thread.java:840)
java.lang.OutOfMemoryError: Java heap space
 Splash Rendering 50 Group running
Waiting for splash to open before moving on
 Splash Rendering 50 To Hole running
Waiting for splash to open before moving on
 Splash Rendering 50 Group running
Waiting for splash to open before moving on
 Splash Rendering 50 Move Center running
Waiting for splash to open before moving on
 Splash Rendering 50 Move Center running
Waiting for splash to open before moving on
java.lang.OutOfMemoryError: Java heap space
java.lang.Exception: UI Thread Exception here!
	at com.neuronrobotics.bowlerstudio.BowlerKernel.runLater(BowlerKernel.java:801)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:447)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:412)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:175)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$29(SelectionSession.java:1167)
	at java.base/java.lang.Thread.run(Thread.java:840)
 Splash Rendering 50 Group running
Waiting for splash to open before moving on
CaDoodle Updater clean exit
```

JansenSmith, 2024-11-23:
```
Pruning polygon loading STL::file
java.lang.OutOfMemoryError: Java heap space
	at eu.mihosoft.vrl.v3d.Vector3d.clone(Vector3d.java:150)
	at eu.mihosoft.vrl.v3d.Vertex.clone(Vertex.java:89)
	at eu.mihosoft.vrl.v3d.Polygon.lambda$clone$0(Polygon.java:217)
	at eu.mihosoft.vrl.v3d.Polygon$$Lambda$311/0x00007d55e8603958.accept(Unknown Source)
	at java.base/java.util.ArrayList.forEach(ArrayList.java:1511)
	at eu.mihosoft.vrl.v3d.Polygon.clone(Polygon.java:216)
	at eu.mihosoft.vrl.v3d.Polygon.transformed(Polygon.java:379)
	at eu.mihosoft.vrl.v3d.CSG.lambda$transformed$8(CSG.java:1942)
	at eu.mihosoft.vrl.v3d.CSG$$Lambda$305/0x00007d55e86034b0.apply(Unknown Source)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1625)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:682)
	at eu.mihosoft.vrl.v3d.CSG.transformed(CSG.java:1948)
	at eu.mihosoft.vrl.v3d.CSG.movez(CSG.java:462)
	at com.neuronrobotics.bowlerstudio.creature.ThumbnailImage.get(ThumbnailImage.java:87)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.lambda$loadingImageFromUIThread$3(CaDoodleFile.java:447)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile$$Lambda$1433/0x00007d55e8a2a178.run(Unknown Source)
	at com.neuronrobotics.bowlerstudio.BowlerKernel.lambda$runLater$7(BowlerKernel.java:815)
	at com.neuronrobotics.bowlerstudio.BowlerKernel$$Lambda$1062/0x00007d55e8865800.run(Unknown Source)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:456)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl$$Lambda$72/0x00007d55e808ecf0.run(Unknown Source)
	at java.base/java.security.AccessController.executePrivileged(AccessController.java:776)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:455)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl$$Lambda$71/0x00007d55e808e880.run(Unknown Source)
	at javafx.graphics/com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$10(GtkApplication.java:264)
java.lang.Exception: UI Thread Exception here!
	at com.neuronrobotics.bowlerstudio.BowlerKernel.runLater(BowlerKernel.java:801)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:447)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:412)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:175)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$29(SelectionSession.java:1167)
	at java.base/java.lang.Thread.run(Thread.java:840)
 Splash Rendering 50 Resize running
Waiting for splash to open before moving on
Waiting for splash to open before moving on
CaDoodle Updater clean exit
```

JansenSmith, 2024-11-23:
```
Waiting for splash to open before moving on
Exception in thread "addOpperation Thread 1" java.lang.OutOfMemoryError: Java heap space
	at eu.mihosoft.vrl.v3d.Vector3d.clone(Vector3d.java:150)
	at eu.mihosoft.vrl.v3d.Vertex.clone(Vertex.java:89)
	at eu.mihosoft.vrl.v3d.Polygon.lambda$clone$0(Polygon.java:217)
	at eu.mihosoft.vrl.v3d.Polygon$$Lambda$322/0x000077c0646117b0.accept(Unknown Source)
	at java.base/java.util.ArrayList.forEach(ArrayList.java:1511)
	at eu.mihosoft.vrl.v3d.Polygon.clone(Polygon.java:216)
	at eu.mihosoft.vrl.v3d.CSG.clone(CSG.java:735)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.storeResultInCache(CaDoodleFile.java:273)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.process(CaDoodleFile.java:213)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.lambda$addOpperation$2(CaDoodleFile.java:241)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile$$Lambda$1456/0x000077c064a1a690.run(Unknown Source)
	at java.base/java.lang.Thread.run(Thread.java:840)
 Splash Rendering 50 To Hole running
Waiting for splash to open before moving on
Exception in thread "addOpperation Thread 1" java.lang.OutOfMemoryError: Java heap space
```

madhephaestus, 2024-11-24:
please do not post out of memory exceptions without a full visualvm graph and memory breakdown. this is a duplicate of #25

closed 2024-11-24 by madhephaestus