CommonWealthRobotics/CaDoodle-Application #38: MACOS mouse unresponsive at loading time

madhephaestus, 2025-01-05:
If you resize the window from its starting size, the mouse starts working.

madhephaestus, 2025-01-07:
this issue looks like its Intel mac only

madhephaestus, 2025-04-17:
As a workaround, try to use the mouse to resize the window slightly. If you drag from a corner to make it slightly smaller, the UI refreshes and sometimes the mouse now interacts with the application window correctly.

samallman87, 2025-08-29:
> As a workaround, try to use the mouse to resize the window slightly. If you drag from a corner to make it slightly smaller, the UI refreshes and sometimes the mouse now interacts with the application window correctly.

This workaround no longer works for me on my intel Mac. Now, about one second after I try resizing the window by dragging a corner, I get a massive kernel panic. No rainbow beach ball or anything, everything just completely stops, including the system clock in the upper right. Screen remains frozen until having to force restart the whole laptop. Have reproduced 4 times consistently without fail, every time same massive kernel panic.  I’m not sure at which version this behavior changed for me

madhephaestus, 2025-08-29:
Whoh, that sucks. 

I immediately pulled out my Intel mac to see if i could reproduce it. I have Macos 13.4.1 one a 2020 Intel Macbook air with 8gb of ram. Unfortunately i was not able to reproduce this error. 

@samallman87 can you run CaDoodle from the Terminal app with:

`/Applications/CaDoodle.app/Contents/MacOS/CaDoodle`

You will get to see all of the print statements. If you see a stack trace just before the freeze, can you take a picture with your phone or something and send it here? 

If you would like to try an older version go to ~/bin/CaDoodle-ApplicationInstall/ and edit currentversion.txt change its contents to what ever older version you have downloaded (see the folders in the same folder as currentversion.txt for your options)  When CaDoodle runs, it will ask you to update, just say no, and it will open the older version.

samallman87, 2025-08-30:
hey, so here's what returns when i launch CaDoodle from the terminal:

Checking CaDoodle-Application.jar
https://github.com/CommonWealthRobotics/CaDoodle-Application/releases/download/0.22.13/CaDoodle-Application.jar Size 422485822 bytes
CaDoodle-Application.jar is not jvm.json
Checking jvm.json
jvm.json is not CaDoodle-Application.jar
https://github.com/CommonWealthRobotics/CaDoodle-Application/releases/download/0.22.13/jvm.json Size 4090 bytes
Not downloadeing, it existst jvm.json
Using JVM /Users/samallman1/bin/CaDoodle-ApplicationInstall/0.22.13/jvm.json
Not downloadeing, it existst zulu17.60.17-ca-fx-jdk17.0.16-macosx_x64.zip
Using JVM /Users/samallman1/bin/CaDoodle-ApplicationInstall/zulu17.60.17-ca-fx-jdk17.0.16-macosx_x64.zip
Not extraction, VM exists /Users/samallman1/bin/CaDoodle-ApplicationInstall/zulu17.60.17-ca-fx-jdk17.0.16-macosx_x64
Running:

/Users/samallman1/bin/CaDoodle-ApplicationInstall/zulu17.60.17-ca-fx-jdk17.0.16-macosx_x64/bin/java -Dprism.forceGPU=true -XX:MaxRAMPercentage=80 --add-exports javafx.graphics/com.sun.javafx.css=ALL-UNNAMED --add-exports javafx.controls/com.sun.javafx.scene.control.behavior=ALL-UNNAMED --add-exports javafx.controls/com.sun.javafx.scene.control=ALL-UNNAMED --add-exports javafx.base/com.sun.javafx.event=ALL-UNNAMED --add-exports javafx.controls/com.sun.javafx.scene.control.skin.resources=ALL-UNNAMED --add-exports javafx.graphics/com.sun.javafx.util=ALL-UNNAMED --add-exports javafx.graphics/com.sun.javafx.scene.input=ALL-UNNAMED --add-opens javafx.graphics/javafx.scene=ALL-UNNAMED  -jar  /Users/samallman1/bin/CaDoodle-ApplicationInstall/0.22.13/CaDoodle-Application.jar

Application at /Users/samallman1/bin/CaDoodle-ApplicationInstall/0.22.13/CaDoodle-Application.jar is Found
 Splash Rendering 1 Main Window Show
Waiting for splash to open before moving on
Loading splash image: jar:file:/Users/samallman1/bin/CaDoodle-ApplicationInstall/0.22.13/CaDoodle-Application.jar!/com/commonwealthrobotics/SourceIcon.png
Waiting for splash to open before moving on
Waiting for splash to open before moving on
Waiting for splash to open before moving on
Waiting for splash to open before moving on
Pervious version was 0.22.13 and current version is 0.22.13
SLF4J: No SLF4J providers were found.
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See https://www.slf4j.org/codes.html#noProviders for further details.
SLF4J: Class path contains SLF4J bindings targeting slf4j-api versions 1.7.x or earlier.
SLF4J: Ignoring binding found at [jar:file:/Users/samallman1/bin/CaDoodle-ApplicationInstall/0.22.13/CaDoodle-Application.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See https://www.slf4j.org/codes.html#ignoredBindings for an explanation.
Getting branches https://github.com/CommonWealthRobotics/BowlerStudioImageAssets.git  Loading asset layout/WebTabLayout.fxml
Getting branches https://github.com/CommonWealthRobotics/BowlerStudioImageAssets.git  Loading asset layout/addFileToGist.fxml
Loading asset layout/CreatureLab.fxml
Loading asset layout/AddRemoveVitamins.fxml
Loading asset layout/flyingCamera.xml
Loading asset layout/findWidget.fxml
Loading asset layout/changeAssetRepo.fxml
Loading asset layout/Main.fxml
Loading asset layout/newCreatureWizard.fxml
Loading asset layout/newCreaturLab.fxml
Loading asset layout/DyIOChannelContorol.fxml
Loading asset layout/jogWidget.fxml
Loading asset layout/jogTrainerWidget.fxml
Loading asset layout/DyIOPanel.fxml
Loading asset layout/CreatureLabControls.fxml
Loading asset layout/release.fxml
Loading asset layout/BowlerConnectionMenu.fxml
Loading asset layout/newVitaminWizard.fxml
Loading asset layout/BowlerStudioMenuBar.fxml
Loading asset layout/githublogin.fxml
Loading asset layout/CreatureLabControlsTab.fxml
Loading asset layout/createNewGist.fxml
Loading asset layout/BowlerStudioModularFrame.fxml
Loading asset layout/Terminal.fxml
Loading asset ground.svg
Loading asset ControllerToolBar.fxml
Closing server connection
 Splash Rendering 50 Loading Shapes
 Splash Rendering 1 Initialize Model
WARNING:This will make STL's incompatible with low quality slicing engines like Slice3r and PrusaSlicer
Advanced mode: false
Using built in link loaders
Setting transparent tray icon to CADoodle-Icon.png
Auto save /Users/samallman1/Library/Application Support/BowlerLauncher/CaDoodle-workspace/MyCaDoodleProjects/Doodle-1-Effervescent_Nanobot/Doodle-1-Effervescent_Nanobot.doodle
Cancel event
microbit value massCentroidX 0.0
microbit value massCentroidY 0.0
microbit value massCentroidZ 0.0
microbit value massKg 0.041
microbit value price 41.09
microbit value source https://shop.elecfreaks.com/products/elecfreaks-micro-bit-wukong-expansion-board-adapter

let me know if you want me to try anything else or need help with anything

vcscsvcscs, 2025-11-06:
Same problem on MacBook Pro M1 32GB, Canvas is fully unresponsive. Everything works except canvas. Doesn't really matter how much time passes it never really loads and everything except canvas is responsive. I am new to Mac. The resize 'hack' solves it, interesting...

vcscsvcscs, 2025-11-06:
> ```As a workaround, try to use the mouse to resize the window slightly. If you drag from a corner to make it slightly smaller, the UI refreshes and sometimes the mouse now interacts with the application window correctly.```
> 
> @vcscsvcscs Did the above workaroud bring it back for you?

Yes

closed 2026-05-03 by madhephaestus