CommonWealthRobotics/CaDoodle-Application #43: Enter does not Re-Generate when entering data into shape

RyMarq, 2025-03-31:
Enter re-generating after entry would be a small but real workflow optimization and improve intuitiveness.

madhephaestus, 2025-03-31:
Are you referring to the parametric options in the context window containing the "Re-Generate" button? 

If so, this workflow was set up specifically because regeneration for some models is a multi-minute process and therefore initiating it should not be the result of normal, knee-jerk actions. Hitting enter sets the value of the field your are editing into the set of staged changes, with the button acting as the initiating event of the parametric regeneration. 

I can look at moving focus to the regenerate button so if you hit enter to set the field, then enter a second time would initiate the regeneration action maybe?

JansenSmith, 2025-03-31:
Thinking it thru, I think TAB should go to the next field and ENTER should skip the rest of the fields and highlight Re-generate. ENTER on re-generate should do the regeneration.

madhephaestus, 2025-03-31:
Can you confirm that is not what it already does? What is the current behavior and how is it different?

JansenSmith, 2025-03-31:
pressing ENTER here sets 2.0 as the value of Round (and so makes Re-Generate selectable) but does NOT auto-select Re-Generate.
In order to select Re-Generate, user must TAB four times. Desired behavior would be for user to hit ENTER twice. First ENTER would set 2.0 as the value of Round and auto-select Re-Generate.
[image]

closed 2025-04-01 by madhephaestus