CommonWealthRobotics/CaDoodle-Application #44: Automated Download Repeating Infinitely.

RyMarq, 2025-04-02:
https://www.7-zip.org/a/7z2407-extra.7z download repeating infinitely until manually stopped.

[image]

RyMarq, 2025-04-02:
Default browser, Firefox with Ublock Origins installed was being called. I don't expect that to matter, but good to mention.

madhephaestus, 2025-04-02:
@RyMarq  i nerfed the 7z system executable use and have it fail back to the pure Java 7z extraction (which is slower). Let me know if it at least is working for you? If not, i would expect to see a download loop on some other plugin, meaning i need to do a deep dive into why Windows is suddenly refusing to allow portable applications from running.

RyMarq, 2025-04-03:
Seems like its working, but the process is a bit buggy. The downloading window is now file-by-file and therefore flickers on and off very rapidly, and is always at 0% (I presume because it finishes basically instantly for each file).

RyMarq, 2025-04-03:
Hm, even after the export process is finished, on the free-cad system, a Zulu window was popping up and closing back and fourth many times per second, when attempting to export to free-cad. Its possible it was still downloading or unpacking, and the window just stop displaying (it started again when I ended the program to reattempt), but the process had taken a half an hour by that stage.

madhephaestus, 2025-04-03:
Hmmm, i just ran through the full workflow in my windows VM and was able to export to Freecad file. 

Try deleting the BowlerStudioInstall and Cadoodle-ApplicationInstall folders found in ~/bin/ then running CaDoodle again. It may have gotten into a state it can not get back out of if it got interrupted during a download at any poiint. If the file its looking for (in this case the 7zip portable application) exists, but is incomplete because of a faulty download, then it tries ot open it, and fails. this exception prompts it to download it again, but this time sends it to the browser to download and for you to manually install, which also seems to fail on windows. 

This may be correctable by changing the DownloadManager to download to a tmp file, then upon a successful download, copies it over to the correct location for extraction... assuming erasing the ~/bin/ folders fixes the problem, that would be the correction needed in the codebase to prevent this error in the future.

madhephaestus, 2025-04-03:
Ok, i made the change where it downloads to a temp file before copy over for the use. This should prevent future problems caused by corrupted downloads. 

You still should delete the folders in `~/bin/`  before trying this.

madhephaestus, 2025-04-04:
I ran a test last night on a Microsoft Surface and was able to verify the downloader was working as expected.

madhephaestus, 2025-04-04:
I added more error checking to the download manager to only attempt a download once per application run.

closed 2025-04-15 by madhephaestus