CommonWealthRobotics/CaDoodle-Application #59: Error - XOR on Scribble

JansenSmith, 2025-06-07:
java.lang.RuntimeException: Failed to create image
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:1001)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.setSaveImage(CaDoodleFile.java:924)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:881)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:370)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$41(SelectionSession.java:1563)
	at java.base/java.lang.Thread.run(Thread.java:840)
Selection Box Background Click 
Cancel event
java.lang.RuntimeException: Failed to create image
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:1001)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.setSaveImage(CaDoodleFile.java:924)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:881)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:370)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$41(SelectionSession.java:1563)
	at java.base/java.lang.Thread.run(Thread.java:840)
java.lang.RuntimeException: Can not be null
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:980)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.setSaveImage(CaDoodleFile.java:924)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:881)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:370)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$41(SelectionSession.java:1563)
	at java.base/java.lang.Thread.run(Thread.java:840)
Auto save /home/jansen/Documents/CaDoodle-workspace/MyCaDoodleProjects/Doodle-36-Profuse_Hydra/Doodle-36-Profuse_Hydra.doodle
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  26.7638533034 0.0000000000 1.0000000000 ,  24.0000000000 0.0000000000 1.0000000000 ,  25.0000000000 0.0000000000 0.0000000000 ,  26.7638533034 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  26.8254578165 0.0000000000 1.0000000000 ,  26.7638533034 0.0000000000 1.0000000000 ,  26.7638533034 0.0000000000 0.0000000000 ,  26.8254578165 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  26.8733108025 0.0000000000 1.0000000000 ,  26.8254578165 0.0000000000 1.0000000000 ,  26.8254578165 0.0000000000 0.0000000000 ,  26.8733108025 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  26.9506149826 0.0000000000 1.0000000000 ,  26.8733108025 0.0000000000 1.0000000000 ,  26.8733108025 0.0000000000 0.0000000000 ,  26.9506149826 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  27.7064933714 0.0000000000 1.0000000000 ,  26.9506149826 0.0000000000 1.0000000000 ,  26.9506149826 0.0000000000 0.0000000000 ,  27.7064933714 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  28.1140553443 0.0000000000 1.0000000000 ,  27.7064933714 0.0000000000 1.0000000000 ,  27.7064933714 0.0000000000 0.0000000000 ,  28.1140553443 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  28.5496151973 0.0000000000 1.0000000000 ,  28.1140553443 0.0000000000 1.0000000000 ,  28.1140553443 0.0000000000 0.0000000000 ,  28.5496151973 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  29.0545891032 0.0000000000 1.0000000000 ,  28.5496151973 0.0000000000 1.0000000000 ,  28.5496151973 0.0000000000 0.0000000000 ,  29.0545891032 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  29.6981850051 0.0000000000 1.0000000000 ,  29.0545891032 0.0000000000 1.0000000000 ,  29.0545891032 0.0000000000 0.0000000000 ,  29.6981850051 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  30.6209776941 0.0000000000 1.0000000000 ,  29.6981850051 0.0000000000 1.0000000000 ,  29.6981850051 0.0000000000 0.0000000000 ,  30.6209776941 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  32.1812726468 0.0000000000 1.0000000000 ,  30.6209776941 0.0000000000 1.0000000000 ,  30.6209776941 0.0000000000 0.0000000000 ,  32.1812726468 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  34.5072483161 0.0000000000 1.0000000000 ,  32.1812726468 0.0000000000 1.0000000000 ,  32.1812726468 0.0000000000 0.0000000000 ,  34.5072483161 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  34.7750107231 0.0000000000 1.0000000000 ,  34.5072483161 0.0000000000 1.0000000000 ,  34.5072483161 0.0000000000 0.0000000000 ,  34.7750107231 0.0000000000 0.0000000000 ,  ] 
Polygon is colinear, removing # points=4 normal=0.0000000000 -1.0000000000 0.0000000000 [  43.7740000000 0.0000000000 1.0000000000 ,  34.7750107231 0.0000000000 1.0000000000 ,  34.7750107231 0.0000000000 0.0000000000 ,  43.7740000000 0.0000000000 0.0000000000 ,  ] 
Pruned 14 polygons from CSG Perky_Drone_h9iCZdf7BP_1
Z distance = 0.75
Z distance = 0.75
java.lang.RuntimeException: Failed to create image
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:1001)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.setSaveImage(CaDoodleFile.java:924)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:881)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:370)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$41(SelectionSession.java:1563)
	at java.base/java.lang.Thread.run(Thread.java:840)
java.lang.RuntimeException: Failed to create image
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:1001)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.setSaveImage(CaDoodleFile.java:924)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:881)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:370)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$41(SelectionSession.java:1563)
	at java.base/java.lang.Thread.run(Thread.java:840)
java.lang.RuntimeException: Failed to create image
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:1001)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.setSaveImage(CaDoodleFile.java:924)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:881)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:370)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$41(SelectionSession.java:1563)
	at java.base/java.lang.Thread.run(Thread.java:840)
java.lang.RuntimeException: Failed to create image
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:1001)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.setSaveImage(CaDoodleFile.java:924)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:881)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:370)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$41(SelectionSession.java:1563)
	at java.base/java.lang.Thread.run(Thread.java:840)
java.lang.RuntimeException: Can not be null
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.loadingImageFromUIThread(CaDoodleFile.java:980)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.setSaveImage(CaDoodleFile.java:924)
	at com.neuronrobotics.bowlerstudio.scripting.cadoodle.CaDoodleFile.save(CaDoodleFile.java:881)
	at com.commonwealthrobotics.ActiveProject.save(ActiveProject.java:370)
	at com.commonwealthrobotics.controls.SelectionSession.lambda$save$41(SelectionSession.java:1563)
	at java.base/java.lang.Thread.run(Thread.java:840)
Select Box Started x=-379.3130625411525 y=-76.24383166656332

madhephaestus, 2025-06-07:
> to @madhephaestus <https://github.com/madhephaestus>.
> <CommonWealthRobotics/CaDoodle-Application/issue/59/issue_event/18036264262
> @github.com>

Can you send the zip?

JansenSmith, 2025-06-07:
@madhephaestus https://drive.google.com/file/d/1XvRsRZlR0YxPeh4srZieDW--lpaW6iCb/view?usp=drive_link

JansenSmith, 2025-06-07:
@madhephaestus interesting....... it's specifically the rose geometries. i just did CTRL+D on the red roses (not the thorns), and it happened again