CommonWealthRobotics/CaDoodle-Application #67: CaDoodle eats all memory on pc shortly after launch.

LeadDennis, 2025-08-09:
CaDoodle eats all memory on pc shortly after launch. 

I was able to create a quick project after installing but now all I get it a huge memory leak. Attached are my projects. 

[MyCaDoodleProjects.zip](https://github.com/user-attachments/files/21702568/MyCaDoodleProjects.zip)

madhephaestus, 2025-08-10:
Hmm, I can not reporduce that error on my Ubuntu development machine, i wonder if it is a Windows specific issue.  Can you provide some information on the windows machine? I wonder if the one-time security scanning is coming into play here. Can you just try it again and see if the error is repeatable?

madhephaestus, 2025-08-12:
I wonder if this is an OpenCL issue. There is GPU acceleration for CSG operations over 10k polygons. I wonder if the windows implementation has a memory leak...

Can you tell me what OpenCL version is running on that computer?

madhephaestus, 2025-08-12:
I made a new release that disables OpenCL on WIndows for the split polygon algo, the only place there is windows specific code at startup. Let me know if the latest is any different?

LeadDennis, 2025-08-12:
Updated at first launch, program took around 5GB of memory to launch first run then choked out. Second launch it is running well and using around 2.6GB of memory looks like it is fixed. Thank you.

closed 2025-08-13 by LeadDennis