CommonWealthRobotics/CaDoodle-Application #68: CaDoodle on mint updates then fails to launch.

LeadDennis, 2025-08-09:
deleted file in bin directory and tried app again to no avail, reinstalled package to no avail. 

Launching the appimage fails as well after the auto updater. 

Checking CaDoodle-Application.jar
https://github.com/CommonWealthRobotics/CaDoodle-Application/releases/download/0.20.6/CaDoodle-Application.jar Size 422414541 bytes
CaDoodle-Application.jar is not jvm.json
Checking jvm.json
jvm.json is not CaDoodle-Application.jar
https://github.com/CommonWealthRobotics/CaDoodle-Application/releases/download/0.20.6/jvm.json Size 4099 bytes
Not downloadeing, it existst jvm.json
Using JVM /home/user/bin/CaDoodle-ApplicationInstall/0.20.6/jvm.json
Not downloadeing, it existst zulu17.50.19-ca-fx-jdk17.0.11-linux_x64.tar.gz
Using JVM /home/user/bin/CaDoodle-ApplicationInstall/zulu17.50.19-ca-fx-jdk17.0.11-linux_x64.tar.gz
Not extraction, VM exists /home/user/bin/CaDoodle-ApplicationInstall/zulu17.50.19-ca-fx-jdk17.0.11-linux_x64
Running:

/home/user/bin/CaDoodle-ApplicationInstall/zulu17.50.19-ca-fx-jdk17.0.11-linux_x64/bin/java -Dprism.forceGPU=true -XX:MaxRAMPercentage=90.0 --add-exports javafx.graphics/com.sun.javafx.css=ALL-UNNAMED --add-exports javafx.controls/com.sun.javafx.scene.control.behavior=ALL-UNNAMED --add-exports javafx.controls/com.sun.javafx.scene.control=ALL-UNNAMED --add-exports javafx.base/com.sun.javafx.event=ALL-UNNAMED --add-exports javafx.controls/com.sun.javafx.scene.control.skin.resources=ALL-UNNAMED --add-exports javafx.graphics/com.sun.javafx.util=ALL-UNNAMED --add-exports javafx.graphics/com.sun.javafx.scene.input=ALL-UNNAMED --add-opens javafx.graphics/javafx.scene=ALL-UNNAMED  -jar  /home/user/bin/CaDoodle-ApplicationInstall/0.20.6/CaDoodle-Application.jar

CaDoodle Updater clean exit

System:
  Kernel: 6.8.0-71-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.4.8 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Machine:
  Type: Laptop System: LENOVO product: 20S6001WUS v: ThinkPad T15 Gen 1
    serial: <superuser required> Chassis: type: 10 serial: <superuser required>
  Mobo: LENOVO model: 20S6001WUS v: SDK0J40697 WIN serial: <superuser required>
    part-nu: LENOVO_MT_20S6_BU_Think_FM_ThinkPad T15 Gen 1 uuid: <superuser required> UEFI: LENOVO
    v: N2XET29W (1.19 ) date: 03/23/2021
Battery:
  ID-1: BAT0 charge: 55.3 Wh (95.2%) condition: 58.1/57.0 Wh (101.9%) power: 6.4 W volts: 12.4
    min: 11.6 model: LGC 5B10W13913 type: Li-poly serial: <filter> status: discharging cycles: 84
CPU:
  Info: quad core model: Intel Core i7-10610U bits: 64 type: MT MCP smt: enabled
    arch: Comet/Whiskey Lake note: check rev: C cache: L1: 256 KiB L2: 1024 KiB L3: 8 MiB
  Speed (MHz): avg: 1900 high: 1903 min/max: 400/4900 cores: 1: 1900 2: 1903 3: 1901 4: 1900
    5: 1901 6: 1900 7: 1900 8: 1900 bogomips: 36799
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel CometLake-U GT2 [UHD Graphics] vendor: Lenovo driver: i915 v: kernel
    arch: Gen-9.5 ports: active: eDP-1 empty: DP-1, DP-2, HDMI-A-1, HDMI-A-2 bus-ID: 00:02.0
    chip-ID: 8086:9b41 class-ID: 0300
  Device-2: Chicony Integrated Camera driver: uvcvideo type: USB rev: 2.0 speed: 480 Mb/s
    lanes: 1 bus-ID: 1-8:3 chip-ID: 04f2:b6d0 class-ID: fe01 serial: <filter>
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: modesetting
    unloaded: fbdev,vesa dri: iris gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96 s-size: 508x285mm (20.00x11.22") s-diag: 582mm (22.93")
  Monitor-1: eDP-1 model: BOE Display 0x08d8 res: 1920x1080 hz: 60 dpi: 142
    size: 344x194mm (13.54x7.64") diag: 395mm (15.5") modes: 1920x1080
  API: EGL v: 1.5 hw: drv: intel iris platforms: device: 0 drv: iris device: 1 drv: swrast gbm:
    drv: iris surfaceless: drv: iris x11: drv: iris inactive: wayland
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: intel mesa v: 25.0.7-0ubuntu0.24.04.1 glx-v: 1.4
    direct-render: yes renderer: Mesa Intel UHD Graphics (CML GT2) device-ID: 8086:9b41
Audio:
  Device-1: Intel Comet Lake PCH-LP cAVS vendor: Lenovo driver: sof-audio-pci-intel-cnl
    bus-ID: 00:1f.3 chip-ID: 8086:02c8 class-ID: 0403
  API: ALSA v: k6.8.0-71-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
Network:
  Device-1: Intel Comet Lake PCH-LP CNVi WiFi driver: iwlwifi v: kernel bus-ID: 00:14.3
    chip-ID: 8086:02f0 class-ID: 0280
  IF: wlp0s20f3 state: up mac: <filter>
  Device-2: Intel Ethernet I219-LM vendor: Lenovo driver: e1000e v: kernel port: N/A
    bus-ID: 00:1f.6 chip-ID: 8086:0d4e class-ID: 0200
  IF: enp0s31f6 state: down mac: <filter>
Bluetooth:
  Device-1: Intel AX201 Bluetooth driver: btusb v: 0.8 type: USB rev: 2.0 speed: 12 Mb/s lanes: 1
    bus-ID: 1-10:5 chip-ID: 8087:0026 class-ID: e001
  Report: hciconfig ID: hci0 rfk-id: 1 state: up address: <filter> bt-v: 5.2 lmp-v: 11
    sub-v: 20f9 hci-v: 11 rev: 20f9 class-ID: 7c010c
Drives:
  Local Storage: total: 116.06 GiB used: 68.93 GiB (59.4%)
  ID-1: /dev/sda vendor: Lexar model: USB Flash Drive size: 116.06 GiB type: USB rev: 3.2
    spd: 5 Gb/s lanes: 1 tech: SSD serial: <filter> fw-rev: 1.00 scheme: GPT
Partition:
  ID-1: / size: 113.19 GiB used: 68.93 GiB (60.9%) fs: ext4 dev: /dev/sda2
  ID-2: /boot/efi size: 511 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/sda1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
USB:
  Hub-1: 1-0:1 info: hi-speed hub with single TT ports: 12 rev: 2.0 speed: 480 Mb/s lanes: 1
    chip-ID: 1d6b:0002 class-ID: 0900
  Device-1: 1-6:2 info: G2Touch Multi-Touch by G2TSP type: HID
    driver: hid-generic,hid-multitouch,usbhid interfaces: 2 rev: 2.0 speed: 12 Mb/s lanes: 1
    power: 100mA chip-ID: 2a94:564d class-ID: 0300
  Device-2: 1-8:3 info: Chicony Integrated Camera type: video driver: uvcvideo interfaces: 5
    rev: 2.0 speed: 480 Mb/s lanes: 1 power: 500mA chip-ID: 04f2:b6d0 class-ID: fe01 serial: <filter>
  Device-3: 1-9:4 info: Synaptics Prometheus MIS Touch Fingerprint Reader type: <vendor specific>
    driver: N/A interfaces: 1 rev: 2.0 speed: 12 Mb/s lanes: 1 power: 100mA chip-ID: 06cb:00bd
    class-ID: ff00 serial: <filter>
  Device-4: 1-10:5 info: Intel AX201 Bluetooth type: bluetooth driver: btusb interfaces: 2
    rev: 2.0 speed: 12 Mb/s lanes: 1 power: 100mA chip-ID: 8087:0026 class-ID: e001
  Hub-2: 2-0:1 info: super-speed hub ports: 6 rev: 3.1 speed: 10 Gb/s lanes: 1 chip-ID: 1d6b:0003
    class-ID: 0900
  Device-1: 2-4:2 info: Lexar USB Flash Drive type: mass storage driver: usb-storage
    interfaces: 1 rev: 3.2 speed: 5 Gb/s lanes: 1 power: 400mA chip-ID: 21c4:0cd1 class-ID: 0806
    serial: <filter>
  Hub-3: 3-0:1 info: hi-speed hub with single TT ports: 2 rev: 2.0 speed: 480 Mb/s lanes: 1
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-4: 4-0:1 info: super-speed hub ports: 2 rev: 3.1 speed: 10 Gb/s lanes: 1 chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  System Temperatures: cpu: 44.0 C mobo: N/A
  Fan Speeds (rpm): fan-1: 0
Repos:
  Packages: 2412 pm: dpkg pkgs: 2397 pm: flatpak pkgs: 15
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com xia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu noble main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu noble-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu noble-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ noble-security main restricted universe multiverse
Info:
  Memory: total: 16 GiB note: est. available: 15.25 GiB used: 2.36 GiB (15.5%)
  Processes: 283 Power: uptime: 11m states: freeze,mem,disk suspend: s2idle wakeups: 0
    hibernate: platform Init: systemd v: 255 target: graphical (5) default: graphical
  Compilers: gcc: 13.3.0 Client: Cinnamon v: 6.4.8 inxi: 3.3.34

madhephaestus, 2025-08-12:
CaDoodle downloads and runs a stand-alone JVM into `/home/user/bin/CaDoodle-ApplicationInstall/zulu17.50.19-ca-fx-jdk17.0.11-linux_x64/bin/java` 

can you run `/home/user/bin/CaDoodle-ApplicationInstall/zulu17.50.19-ca-fx-jdk17.0.11-linux_x64/bin/java --version` to see if the stand alone java is working? Try deleting `/home/user/bin/CaDoodle-ApplicationInstall`  and running it again? 

I will go and get the latest JVM and set it as the next release in case there is some linked library that has been obsoleted in Mint?

LeadDennis, 2025-08-12:
Running /home/user/bin/CaDoodle-ApplicationInstall/zulu17.50.19-ca-fx-jdk17.0.11-linux_x64/bin/java --version give me the same could not find libjava.so could not find the java se runtime environment. 

Auto updater fixed the issue though. Nice job thank you